NordicSemiconductor/pc-ble-driver-js #12: Does this work with nRF52 ?

ksachdeva, 2016-03-21:
Hi,

Would this project work with nRF52 as well ?. The connectivity app seems to be compiled only for nRF51 target.

Regards & thanks
Kapil

kenr, 2016-03-22:
Hi,

We are currently working nRF52 support. You can follow the progress in the update-driver-sd2.0 branch.

BR,
-Ken.

ksachdeva, 2016-03-22:
Thanks Ken. I will follow the branch. A quick look is showing that the driver is still that for nrf51. May be it is coming soon as you mentioned.

Regard
Kapil

kenr, 2016-04-09:
Hi again, we have connectivity firmware for nRF52 in the update-driver-sd2.0 branch that is quite stable; https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/update-driver-sd2.0/driver/hex/connectivity_115k2_with_s132_2.0.0.hex

closed 2016-04-09 by kenr