NordicSemiconductor/pc-ble-driver-js #15: Connectivity firmware

rosek86, 2016-04-11:
Hi,

Where can I find working copy of connectivity_115k2_with_s132_2.0.0 source code, please? I've been trying to use the one from SDK 11 (ble_central_and_peripheral\ble_connectivity\pca10040), both UART and HCI version but this seems to file on

    #2 1 logMessage:        5 ->  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
    #2 1 logMessage: State change: STATE_UNINITIALIZED -> STATE_FAILED

I'd like to add simple shockburst support to connectivity firmware.

Regards,
Krzysztof

kenr, 2016-04-11:
Hi,

Use the nRF5 SDK (v11) and apply the patch that I pushed just now:
https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/update-driver-sd2.0/driver/hex/SD20.patch.

For the patch to succeed you must ignore whitespaces: --ignore-whitespace

You have to set the HCI_LINK_CONTROL define before compiling.

BR,
-Ken.

rosek86, 2016-04-11:
@kenr Thank you for the prompt response! 

The patch worked very well and I can move forward with my task! :-)

BR,
K.

closed 2016-04-11 by rosek86

rosek86, 2016-04-11:
--- components/serialization/connectivity/codecs/s130/middleware/conn_mw_items.c
    +++ components/serialization/connectivity/codecs/s130/middleware/conn_mw_items.c
    @@ -80,4 +80,5 @@
         {SD_BLE_GATTS_RW_AUTHORIZE_REPLY, conn_mw_ble_gatts_rw_authorize_reply},
         {SD_BLE_GATTS_SYS_ATTR_SET, conn_mw_ble_gatts_sys_attr_set},
         {SD_BLE_GATTS_SYS_ATTR_GET, conn_mw_ble_gatts_sys_attr_get},
    +    {SD_BLE_USER_MEM_REPLY, conn_mw_ble_user_mem_reply},
    };

It's not an issue but do you need to add SD_BLE_USER_MEM_REPLY? It seems that this command is already in line 23. The SD_BLE_GAP_SEC_INFO_REPLY is doubled in conn_mw_items.c file too.