NordicSemiconductor/pc-ble-driver-js #16: driver.open is not a function ... error when running the example_scan.js

ksachdeva, 2016-04-18:
Hi,

I am able to buid the "master" branch on OS X successfully however when I run 

node examples/example_scan.js I am getting below mentioned error

```
  driver.open(
           ^

TypeError: driver.open is not a function
    at /Users/ksachdeva/Desktop/Dev/exp/bleexp/pc-ble-driver-js/examples/example_scan.js:33:12

```

I did print out driver object and do not see 'open' function on it.

Please guide.

Regards
Kapil

kenr, 2016-04-19:
Hi, we've not had time to clean up tests/examples properly yet. I've done some quick cleanup now and ran the tests and they pass. Please let me know if you find more more issues with the tests :-)

ksachdeva, 2016-04-19:
Thanks for the quick response. Able to finally run the firmware. Like the stuff you guys are doing. keep up the good work.

closed 2016-04-19 by ksachdeva