NordicSemiconductor/pc-ble-driver-js #17: Example of how to connect to a peripheral and read the characterisitic

ksachdeva, 2016-04-20:
Hi,

Would it be possible to have an example showing how to read the characteristic from a peripheral device ?

Regards & thanks
Kapil

kenr, 2016-04-21:
Hi, currently we do not have any examples showing how to read the characteristics, but we use this part of the API in a large application we are working on (https://devzone.nordicsemi.com/blogs/801/alpha-090-release-of-new-bluetooth-smart-tool-from/). When the application is released we will improve and expand examples and documentation for the API.

ksachdeva, 2016-06-30:
Hi,

Created an example repository showing how to use pc-ble-driver-js

https://github.com/ksachdeva/pc-ble-driver-js-examples

Regards
Kapil

closed 2016-06-30 by ksachdeva

bihanssen, 2016-07-01:
Hi, that's a great initiative. Thanks for sharing.