NordicSemiconductor/pc-ble-driver-js #18: Possible race condition issue in h5_transport

rosek86, 2016-05-10:
Hi,

Occasionally the application doesn't change state from STATE_START to STATE_RESET. I think the reason is that the _exitCriterias->isOpened is set to true before state machine is executed: 

https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/master/driver/src/transport/h5_transport.cpp#L136

If this is the case, the exit->reset() clears isOpened flag and application stucks in "while (!exit->isFullfilled())" loop.

https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/master/driver/src/transport/h5_transport.cpp#L434

BR,
Krzysztof

bihanssen, 2016-05-11:
Thanks for the detailed observation, we'll check it out.

rosek86, 2016-06-15:
I think this was related to my issue with Intel Edison #22, which is now solved. Thanks!

closed 2016-06-15 by rosek86