NordicSemiconductor/pc-ble-driver-js #22: Intel Edison application start issue

rosek86, 2016-06-14:
Hi,

I'm trying to run pc-ble-driver-js on the Intel Edison. The driver compiles without any problem but when I try to run test application starts very occasionally. Usually it fails like this:

> onAdded: Adapter added. Adapter: 000682039986 +235ms
> #1 stateChanged: {"_instanceId":"000682039986./dev/ttyACM0","_port":"/dev/ttyACM0","_serialNumber":"000682039986","baudRate":115200,"parity":"none","flowControl":"none","available":false,"scanning":false,"advertising":false,"connecting":false,"_address":null,"_addressType":null,"name":null,"firmwareVersion":null} +64ms
> #1 logMessage: Successfully opened /dev/ttyACM0. Baud rate: 115200. Flow control: none. Parity: none.
>  +17ms
> 
> Failed to open the nRF5 BLE driver.
> node: ../deps/uv/src/unix/core.c:165: uv_close: Assertion `0' failed.
> Aborted

or 

> onAdded: Adapter added. Adapter: 000682039986 +241ms
> #1 stateChanged: {"_instanceId":"000682039986./dev/ttyACM0","_port":"/dev/ttyACM0","_serialNumber":"000682039986","baudRate":115200,"parity":"none","flowControl":"none","available":false,"scanning":false,"advertising":false,"connecting":false,"_address":null,"_addressType":null,"name":null,"firmwareVersion":null} +123ms
> #1 logMessage: Successfully opened /dev/ttyACM0. Baud rate: 115200. Flow control: none. Parity: none. +46ms
> #1 logMessage:     1/ 0 <-  [01 7e ]
>           type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
>           [SYNC] +81ms
> #1 logMessage:     2/ 0 <-  [01 7e ]
>           type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
>           [SYNC] +345ms
> #1 logMessage:     3/ 0 <-  [01 7e ]
>           type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
>           [SYNC] +340ms
> #1 logMessage:     4/ 0 <-  [01 7e ]
>           type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
>           [SYNC] +345ms
> #1 logMessage:     5/ 0 <-  [01 7e ]
>           type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
>           [SYNC] +340ms
> #1 logMessage:     6/ 0 <-  [01 7e ]
>           type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
>           [SYNC] +345ms
> 
> Failed to open the nRF5 BLE driver.
> node: ../deps/uv/src/unix/core.c:165: uv_close: Assertion `0' failed.
> Aborted

Can you please advice what is wrong here? I'm using standard nRF52 development kit with 2.0.1. hex firmware connected directly to the USB port on the Intel Edison dev kit. The pc-ble-driver-js and pc-ble-driver are both in the newest version.

Many thanks,
Krzysztof

kenr, 2016-06-15:
Hi,

Have you flashed this hex file on the DK ? https://github.com/NordicSemiconductor/pc-ble-driver/blob/8a5cb05b459f3f6de9519f8dee06cb407933abf3/hex/connectivity_115k2_with_s132_2.0.1.hex

BR,
-Ken.

rosek86, 2016-06-15:
Thanks for speedy reply.

1) Yes, that's right. I was initially using my own compilation (using your patch) with a few modifications to add half duplex shock burst support over SPI but had similar issues with that firmware too. I'm now moving back to the original firmware to eliminate my problems step by step.

2) I don't think there are other scripts running at the same time, but I'll confirm today to make sure.

3) Yes, I'll send an example later today.

Regards,
Krzysztof

rosek86, 2016-06-15:
Hey,

I've actually tried to use the newest Intel Edison image with 'new' kernel (released a week ago) and my application runs correctly each time now. I'm not sure what was the issue, maybe a bug in very old kernel?

Many thanks!

closed 2016-06-15 by rosek86