NordicSemiconductor/pc-ble-driver-js #23: Build on OS X is failing

ksachdeva, 2016-06-29:
Please find here the log

info DIST Downloading distribution files.
http DIST 	- http://nodejs.org/dist/v5.11.0/SHASUMS256.txt
http DIST 	- http://nodejs.org/dist/v5.11.0/node-v5.11.0-headers.tar.gz
info CMD CLEAN
info RUN cmake -E remove_directory "/Users/ksachdeva/Desktop/Dev/exp/nordic/pc-ble-driver-js/build"
info CMD CONFIGURE
info RUN cmake "/Users/ksachdeva/Desktop/Dev/exp/nordic/pc-ble-driver-js" --no-warn-unused-cli -GNinja -DCMAKE_BUILD_TYPE="Release" -DCMAKE_LIBRARY_OUTPUT_DIRECTORY="/Users/ksachdeva/Desktop/Dev/exp/nordic/pc-ble-driver-js/build/Release" -DCMAKE_JS_INC="/Users/ksachdeva/.cmake-js/node-x64/v5.11.0/include/node;/Users/ksachdeva/Desktop/Dev/exp/nordic/pc-ble-driver-js/node_modules/nan" -DNODE_RUNTIME="node" -DNODE_RUNTIMEVERSION="5.11.0" -DNODE_ARCH="x64" -DCMAKE_C_COMPILER="clang" -DCMAKE_CXX_COMPILER="clang++" -DCMAKE_CXX_FLAGS="-D_DARWIN_USE_64_BIT_INODE=1 -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -DBUILDING_NODE_EXTENSION -w -std=c++11" -DCMAKE_SHARED_LINKER_FLAGS="-undefined dynamic_lookup"
Not searching for unused variables given on the command line.
-- The C compiler identification is AppleClang 7.3.0.7030031
-- The CXX compiler identification is AppleClang 7.3.0.7030031
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Architecture not set, using native 64-bit toolchain.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE
-- Boost version: 1.60.0
-- Found the following Boost libraries:
--   thread
--   system
--   regex
--   date_time
--   chrono
--   atomic
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/ksachdeva/Desktop/Dev/exp/nordic/pc-ble-driver-js/build
info CMD BUILD
info RUN cmake --build "/Users/ksachdeva/Desktop/Dev/exp/nordic/pc-ble-driver-js/build" --config Release
[99/145] Linking CXX executable pc-ble-driver/test_uart
FAILED: pc-ble-driver/test_uart
: && /usr/bin/clang++   -D_DARWIN_USE_64_BIT_INODE=1 -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -DBUILDING_NODE_EXTENSION -w -std=c++11 -O3 -DNDEBUG -arch x86_64 -arch i386 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names  pc-ble-driver/CMakeFiles/test_uart.dir/test/test_uart.cpp.o  -o pc-ble-driver/test_uart  /usr/local/lib/libboost_thread-mt.a /usr/local/lib/libboost_system-mt.a /usr/local/lib/libboost_regex-mt.a /usr/local/lib/libboost_date_time-mt.a /usr/local/lib/libboost_chrono-mt.a /usr/local/lib/libboost_atomic-mt.a -Wl,-rpath,@loader_path && :
Undefined symbols for architecture i386:
  "boost::detail::thread_data_base::~thread_data_base()", referenced from:
      boost::detail::thread_data<boost::_bi::bind_t<unsigned long, boost::_mfi::mf0<unsigned long, boost::asio::io_service>, boost::_bi::list1<boost::_bi::value<boost::asio::io_service*> > > >::~thread_data() in test_uart.cpp.o
      boost::detail::thread_data<boost::_bi::bind_t<unsigned long, boost::_mfi::mf0<unsigned long, boost::asio::io_service>, boost::_bi::list1<boost::_bi::value<boost::asio::io_service*> > > >::~thread_data() in test_uart.cpp.o
  "boost::system::system_category()", referenced from:
      _main in test_uart.cpp.o
      boost::asio::io_service::run() in test_uart.cpp.o
      boost::asio::detail::service_registry::service_registry<boost::asio::detail::task_io_service, unsigned long>(boost::asio::io_service&, boost::asio::detail::task_io_service*, unsigned long) in test_uart.cpp.o
      boost::asio::detail::task_io_service::~task_io_service() in test_uart.cpp.o
      boost::asio::detail::task_io_service::~task_io_service() in test_uart.cpp.o
      boost::asio::detail::task_io_service::shutdown_service() in test_uart.cpp.o
      boost::asio::detail::task_io_service::run(boost::system::error_code&) in test_uart.cpp.o
      ...
  "boost::system::generic_category()", referenced from:
      __GLOBAL__sub_I_test_uart.cpp in test_uart.cpp.o
  "boost::thread::start_thread_noexcept()", referenced from:
      _main in test_uart.cpp.o
  "boost::thread::detach()", referenced from:
      boost::thread::~thread() in test_uart.cpp.o
  "typeinfo for boost::detail::thread_data_base", referenced from:
      typeinfo for boost::detail::thread_data<boost::_bi::bind_t<unsigned long, boost::_mfi::mf0<unsigned long, boost::asio::io_service>, boost::_bi::list1<boost::_bi::value<boost::asio::io_service*> > > > in test_uart.cpp.o
  "vtable for boost::detail::thread_data_base", referenced from:
      boost::detail::thread_data_base::thread_data_base() in test_uart.cpp.o
  NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
ld: symbol(s) not found for architecture i386
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[104/145] Building CXX object pc-ble-driver/CMakeFiles/pc_ble_driver_obj.dir/src/transport/uart_boost.cpp.o
ninja: build stopped: subcommand failed.
ERR! OMG Process terminated: 1
Unhandled rejection Error: Process terminated: 1
    at ChildProcess.<anonymous> (/Users/ksachdeva/Desktop/Dev/exp/nordic/pc-ble-driver-js/node_modules/cmake-js/lib/cMake.js:372:28)
    at emitTwo (events.js:100:13)
    at ChildProcess.emit (events.js:185:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:204:12)

bihanssen, 2016-06-29:
Hi, 
it seems to me that the compiler is not satisfied with the Boost libraries on your system. If you haven't already, please have a look at [pc-ble-driver repo](https://github.com/NordicSemiconductor/pc-ble-driver#macos-os-x-1011-and-later) on how to build/download Boost.

ksachdeva, 2016-06-29:
brew install boost --universal

I have had boost for some time now. Reinstalled it but still the same error. Then --universal did the trick. 

Thanks.

closed 2016-06-29 by ksachdeva

bihanssen, 2016-06-29:
Thanks for the update.