NordicSemiconductor/pc-ble-driver-js #26: frequent NRF_ERROR_INVALID_STATE error

ksachdeva, 2016-08-09:
Hi,

Here are the steps my application is doing -

a) start scan
b) if interested in device then I stop the scan, connect to device 
c) discover services
d) discover chars
e) read chars
f) disconnect device
g) got to step (a)

Quite often I notice that the callback when doing 'connect to device' does not fire. I then timeout and start the scan process again.

This is when I start to get NRF_ERROR_INVALID_STATE error. Sometimes the my process recover i.e. I am able to successfully do scanning but often I can not recover once I start to get this error.

Based on above few questions:

1) Why the callback for connect is not firing some times ?. 
2) How to recover from NRF_ERROR_INVALID_STATE with out killing the process and/or resetting the device manually ?
3) Is there an API to reset the device ?

Above has become a critical issue & blocker for me. Would sincerely appreciate any guidance on this matter. 

Regards & thanks
Kapil

bihanssen, 2016-09-07:
Hi, sorry for the delay. Are you still seeing the same problem or have you got some new information? 

* When you say you timeout, do you mean you receive the connectTimedOut event of the API?
* The only way to trigger a reset would be to call close/open
* There is a reset in the tranport layer but this is not accessible from the public API

ksachdeva, 2016-09-08:
Hi @bihanssen,

I did manage to solve it. Here is what I did -

To summarize the problem again : Often the callback from 'connect' api will not fire back. 

I added the timeout logic in my application and if I have not received the timeout within required/configured time period then I check the value of adapter.state.checking variable and if it is still true then I invoke cancelConnect api on adapter.

This put the driver (and may be the DK) in the right state.

Regards & thanks
Kapil

bihanssen, 2016-09-09:
Hi,
I found a bug in adapter.js, _parseGapTimeoutEvent function. A fix for this was added in 67063c1. The callback should now be called when the connection procedure times out.

bihanssen, 2016-09-09:
Closing the issue, reopen if there are still problems.

closed 2016-09-09 by bihanssen