NordicSemiconductor/pc-ble-driver-js #28: Stability issues on Ubuntu

ksachdeva, 2016-09-08:
Hi,

My primary development machine has been OS X and things have worked quite well. However my app is to be deployed on Ubuntu and am facing lot of stability issues when opening the adapter.

When adapter.open() is called then a lot of time I end up with below mentioned error message

Failed to open the nRF5 BLE driver.
node: ../deps/uv/src/unix/core.c:168: uv_close: Assertion `0' failed.

Even connecting/disconnecting the DK does not help. Some times rebooting the Ubuntu helps

Any help/guidance on what may be wrong and if any workaround will be very helpful

Regards & thanks
Kapil

bihanssen, 2016-09-09:
Hi,
thanks for reporting.

This sounds like something we will need to look into, we do not currently have any info on this.

bihanssen, 2016-09-13:
Hi,
we found and fixed a bug yesterday that might be related to the issue you were seeing, 17c61b7c48a735f48cda54c0ab38343eebeebe42.
Could you please try with the latest version and see if the issue still remains?

rosek86, 2016-09-13:
Hi @bihanssen,

It seems that your commit fixes that problem on Intel Edison (Yocto).

Many thanks,
Krzysztof

ksachdeva, 2016-09-13:
Hi @bihanssen,

Unfortunately not the good news on my end. The modification helps with a cleaner shut down (earlier it was a crash as shown above) now but has the same/similar problem with a different message (I believe the error is still the same)

Here is what I have done :

- Latest baseline of this repository
- run the simplescan example from this repository
- while in the middle of the scan, do ctrl-c
- re run the simplescan example
- after few tries I starts to get the message below 

Please note :

- removing and re-inserting the nordic DK does not solve the problem. this means that
it is not that connectivity firmware is in some weird state.

My environment :

- Ubuntu 16.04 64-bit
- node 6.5.0
- latest connectivity firmware

###

c: close adapter
onAdded: 000682073463
onAdded: Adapter added. Adapter: 000682073463
s pressed
Running tests
#1 stateChanged: {"_instanceId":"000682073463./dev/ttyACM0","_port":"/dev/ttyACM0","_serialNumber":"000682073463","baudRate":115200,"parity":"none","flowControl":"none","available":false,"scanning":false,"advertising":false,"connecting":false,"_address":null,"_addressType":null,"name":null,"firmwareVersion":null}
#1 logMessage: Successfully opened /dev/ttyACM0. Baud rate: 115200. Flow control: none. Parity: none.

#1 status: {"id":6,"name":"RESET_PERFORMED","message":"Target Reset performed","time":"2016-09-13T13:47:31.902Z"}

Failed to open the nRF5 BLE driver.
#1 error: {
 "message": "Error occurred opening serial port.",
 "description": {
  "errno": 13,
  "errcode": "NRF_ERROR_TIMEOUT",
  "erroperation": "opening port",
  "errmsg": "Error occured when opening port. Errorcode: NRF_ERROR_TIMEOUT (0xd)\n"
 }
}

assert.js:85
  throw new assert.AssertionError({
  ^
AssertionError: false == true
    at adapterOne.open.error (/home/ksachdeva/Desktop/pc-ble-driver-js/test/simpleScanTest.js:102:13)
    at Adapter._checkAndPropagateError (/home/ksachdeva/Desktop/pc-ble-driver-js/api/adapter.js:192:29)
    at _adapter.open.err (/home/ksachdeva/Desktop/pc-ble-driver-js/api/adapter.js:291:22)
➜  pc-ble-driver-js git:(master)

rosek86, 2016-10-26:
Hi @ksachdeva, @bihanssen 

After longer testing time, I've started to get exactly the same issue as @ksachdeva shown above. I investigated this further and it seems to me that this is a race condition problem in h5_transport in the pc-ble-driver.

The H5Transport::open runs UartBoost::open and then sets _exitCriterias->isOpened to true when port is successfully opened. However startStateMachine() starts new thread which sets _exitCriterias->isOpened to false by executing exit->reset() method. If state machine thread runs first everything works well but if H5Transport::open sets isOpened first, the program fails.

I've added a few debug messages and logged program flow for expected and unexpected STATE_START initialization.

Exepcted STATE_START initialization 

> logMessage: STATE_START:init
> logMessage: STATE_START:exit->reset(); // isOpened is set to false
> logMessage: STATE_START:while (!exit->isFullfilled()) {...} // thread waits for isOpened == true
> logMessage: Successfully opened /dev/ttyACM0. Baud rate: 115200. Flow control: none. Parity: none.
> logMessage: _exitCriterias->isOpened = true
> logMessage: waitForState(STATE_ACTIVE, OPEN_WAIT_TIMEOUT)
> logMessage: STATE_START:ok
> logMessage: STATE_RESET:init

Unexpected STATE_START initialization

> logMessage: Successfully opened /dev/ttyACM0. Baud rate: 115200. Flow control: none. Parity: none.
> logMessage: _exitCriterias->isOpened = true
> logMessage: waitForState(STATE_ACTIVE, OPEN_WAIT_TIMEOUT)
> logMessage: STATE_START:init
> logMessage: STATE_START:exit->reset(); // isOpened is set to false
> logMessage: STATE_START:while (!exit->isFullfilled()) {...} // thread waits for isOpened == true

For now I applied a nasty work around as I don't have time to look into this further. I simply added 500 ms delay before setting isOpened to true in H5Transport::open method.

Simply add the following line in L135 of h5_transport.cpp.
`std::this_thread::sleep_for(std::chrono::milliseconds(500));`

https://github.com/NordicSemiconductor/pc-ble-driver/blob/3b805a32e64759c308ad04c0b89faaf82bfed95f/src/transport/h5_transport.cpp#L135

@ksachdeva are you able to check if the work around works for you too? 

I may look into this properly later.

BR,
Krzysztof

bihanssen, 2016-10-28:
@rosek86, thanks for sharing your findings and for the proposed workaround.

vikeri, 2017-06-05:
@rosek86 How do I recompile after I've added your fix?

rosek86, 2017-06-06:
@vikeri there is build.js script in the main directory, you can run 'node build.js'

rosek86, 2017-06-22:
Hi,

I had another problem when setting up the driver on Ubuntu 16.04 LTS. Basically, the driver never enters ACTIVE state. As a workaround, I have increased PACKETS_RETRANSMISSIONS from 4 to 255 and now it enters active time each time.

https://github.com/NordicSemiconductor/pc-ble-driver/blob/4bda53f1ad73b0ba55e02842052e1843c69aea1f/src/common/transport/h5_transport.cpp#L72

kenr, 2018-02-15:
@rosek86 @ksachdeva Are you able to reproduce the error now? There has been some fixes in pc-ble-driver that should fix this issue (https://github.com/NordicSemiconductor/pc-ble-driver/pull/65)