NordicSemiconductor/pc-ble-driver-js #30: Error opening adapter give assert error when cleaning up resources

kenr, 2016-11-15:
Platform: Ubuntu 14.04 LTS (x64)

```[Thread 0x7ffff15c4700 (LWP 29589) exited]
node: ../deps/uv/src/unix/core.c:165: uv_close: Assertion `0' failed.

Program received signal SIGABRT, Aborted.
0x00007ffff6c01c37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56      ../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff6c01c37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff6c05028 in __GI_abort () at abort.c:89
#2  0x00007ffff6bfabf6 in __assert_fail_base (fmt=0x7ffff6d4b3b8 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n",
    assertion=assertion@entry=0x11224f7 "0", file=file@entry=0x12e44e8 "../deps/uv/src/unix/core.c",
    line=line@entry=165, function=function@entry=0x12e4767 "uv_close") at assert.c:92
#3  0x00007ffff6bfaca2 in __GI___assert_fail (assertion=0x11224f7 "0",
    file=0x12e44e8 "../deps/uv/src/unix/core.c", line=165, function=0x12e4767 "uv_close") at assert.c:101
#4  0x0000000000fa4f69 in uv_close ()
#5  0x00007ffff48f767b in Adapter::cleanUpV8Resources() ()
   from /home/kere/dev/cloud-iris/gateway/javascript/console/node_modules/pc-ble-driver-js/build/Release/pc-ble-driver-js.node
#6  0x00007ffff4903cf2 in Adapter::AfterOpen(uv_work_s*) ()
   from /home/kere/dev/cloud-iris/gateway/javascript/console/node_modules/pc-ble-driver-js/build/Release/pc-ble-driver-js.node
#7  0x0000000000fa2755 in uv.work_done ()
#8  0x0000000000fa46ab in ?? ()
#9  0x0000000000fa47a3 in ?? ()
#10 0x0000000000fb4bed in uv.io_poll ()
#11 0x0000000000fa52d6 in uv_run ()
#12 0x0000000000deeea8 in node::Start(int, char**) ()
#13 0x00007ffff6becf45 in __libc_start_main (main=0x722360 <main>, argc=2, argv=0x7fffffffe4b8,
    init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffe4a8)
    at libc-start.c:287
#14 0x00000000007225bf in _start ()
```

vikeri, 2017-06-05:
Also seeing this issue when I try to use an NRF52 DK as adapter. NRF51 DK works as expected

vikeri, 2017-06-05:
I'm on macOS though