NordicSemiconductor/pc-ble-driver-js #31: Internal inconsistency: Could not find device with connection handle 0

rosek86, 2016-11-21:
Hi,

Sometimes when I try to connect to a particular device, the driver stucks in adapter.connect function. There is no connect callback and I don't get connectTimeOut event too.

After a while there is disconnect failed message:

> "message": "Disconnect failed", "description": "Internal inconsistency: Could not find device with connection handle 0"

I tried to add additional timeout for connect method but next time I try to connect to a device this message appears:

> Cannot connect message=Could not connect. Another connect is in progress., description=undefined

Do you have any suggestion how to fix this, please?

bihanssen, 2016-11-21:
Hi, it appears that a Connected event has gone missing. I base this on that you receive a Disconnected event. It's not clear however where the event has gotten lost. What you could do is turn the logLevel option in adapter.Open to debug and collect the info for investigation.

rosek86, 2016-11-22:
Hi,

Please find attached application log, it contains all data from application start to "Disconnect failed" message. The "Status Connecting..." message indicate adapter.connect function.

[application.txt](https://github.com/NordicSemiconductor/pc-ble-driver-js/files/606112/application.txt)

rosek86, 2016-11-22:
Hi again,

Okay, I know what's wrong. My problem is in nan module layer,- there is a 64 slots queue for all events. I have around 15 peripheral devices around, sending advertisement and scan response packets. I'm testing this on Intel Edison which isn't able to get all these events in time before queue is full. That's why connection event is occasionally lost.

I've temporarily increased events queue to 512 slots and this works fine. Much better solution is to have one queue for advertisement/scan response events and second other events.

const auto EVENT_QUEUE_SIZE = 512;
(in adapter.h)

Best regards,
Krix

bihanssen, 2016-11-22:
Thanks for the update! Good catch finding the culprit, we should add a fix for it. Your suggestion about two queues sounds like a good solution.