NordicSemiconductor/pc-ble-driver-js #60: Failed to open the nRF5 BLE driver.

ncperng, 2017-06-08:
Hi,

Thanks for this awesome project. I'm following all the instructions from Installation.md, and successfully built with 'npm install' and passed the 'npm test'. However, while trying the examples I met the following "Failed to open the nRF5 BLE driver" problem as attached.

Should I copy the compiled driver binaries to any other location so that the example program can find them? Thanks for your help in advance.

I'm using OSX 10.12.5, node v7.1.0, and nRF51 dongle nRF6825.
The $DYLD_LIBRARY_PATH is also set to build/Release.

===
$ node examples/heart_rate_monitor.js 
Searching for connected adapters...
onAdded: Adapter added: 000680499321.
Found the following adapters: 
000680499321
Connected to adapter: 000680499321.
Opening adapter with ID: 000680499321 and baud rate: 115200...
#BLE_PERIPHERAL:  logMessage: Successfully opened /dev/cu.usbmodem1411. Baud rate: 115200. Flow control: none. Parity: none.
.
#BLE_PERIPHERAL:  status: {
 "id": 6,
 "name": "RESET_PERFORMED",
 "message": "Target Reset performed",
 "time": "2017-06-08T18:04:36.249Z"
}.

Failed to open the nRF5 BLE driver.
Assertion failed: (0), function uv_close, file ../deps/uv/src/unix/core.c, line 166.
Abort trap: 6

mrodem, 2017-06-09:
Hi! Have you programmed the dongle with the connectivity firmware? For this example to work with nRF51, you need `pc-ble-driver/hex/sd_api_v2/connectivity_1.1.0_115k2_with_s130_2.0.1.hex`.

With nrfjprog:

```
# Erase current firmware
nrfjprog -e -s <serialnumber>

# Program with connectivity firmware
nrfjprog --program connectivity_1.1.0_115k2_with_s130_2.0.1.hex -s <serialnumber>
```

Then run the example. Setting the library path variable should not be necessary.

ncperng, 2017-06-09:
Thank you, Mrodem. 

I could only find connectivity_115k2_with_s130_1.0.0.hex in nrf51-ble-driver_osx_0.5.0.tar.gz. After program it to nrf51 dongle with nrfjprog and run example/heart_rate_monitor.js, same error message here.

However, I download the nRF Connect application and tried to upload firmware. It works.

It is really strange but fortunately I got a workable solution. Will keep testing and deeply appreciated your help.

closed 2017-06-09 by ncperng