NordicSemiconductor/pc-ble-driver-js #64: Failed to request ATT MTU when doing DFU

vikeri, 2017-07-26:
Trying to do a DFU but getting the following error:

```
{ Error: Tried to set ATT MTU 247 for target address C5:22:D7:91:60:59, but got error: Failed to request att mtu
    at createError (/Users/viktor/www/projects/firmware/tools/dfu-server/pc-ble-driver-js/api/dfu/dfuConstants.js:94:19)
    at _adapter.requestAttMtu (/Users/viktor/www/projects/firmware/tools/dfu-server/pc-ble-driver-js/api/dfu/bleTransport.js:424:28)
    at _adapter.gattcExchangeMtuRequest.err (/Users/viktor/www/projects/firmware/tools/dfu-server/pc-ble-driver-js/api/adapter.js:2594:31) code: 21 }
```

It works int the Nordic Connect App.

vikeri, 2017-07-26:
Fixed it by emulating the following settings: [dfu-test.js
](https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/749089aa7c0cfbb240adc134fb0844c0fd8f3ae3/test/jest/dfu-test.js#L239-L244) (but with another baud rate)

mrodem, 2017-09-05:
Issue solved by user. Closing.

closed 2017-09-05 by mrodem