NordicSemiconductor/pc-ble-driver-js #65: Buttonless DFU not working

vikeri, 2017-07-26:
It seems there are some special settings for doing Buttonless DFU, I got the following error:
```
{ Error: Unable to find characteristic 8EC90001F3154F609FB8838830DAEA50 for service FE59 on device C5:22:D7:91:60:58.0
    at createError (/Users/viktor/www/projects/firmware/tools/dfu-server/pc-ble-driver-js/api/dfu/dfuConstants.js:94:19)
    at DeviceInfoService._getCharacteristicIdFromService (/Users/viktor/www/projects/firmware/tools/dfu-server/pc-ble-driver-js/api/dfu/bleTransport/deviceInfoService.js:105:35)
    at _getAttributes.then.then.service (/Users/viktor/www/projects/firmware/tools/dfu-server/pc-ble-driver-js/api/dfu/bleTransport/deviceInfoService.js:69:35)
    at <anonymous> code: 8 }
```
Normal DFU works and buttonless works if I manually write to the descriptor to put the device in DFU mode.

terjesc, 2017-08-21:
Can you check to see if the UUID for the buttonless characteristicis is either 8EC90003F3154F609FB8838830DAEA50 (unbonded) or 8EC90004F3154F609FB8838830DAEA50 (bonded)? If the DFU library does not find any buttonless characteristics it will look for the DFU control point, which is the characteristic reported not to be found in the error that you printed.

If you use log level 'debug' you will get printouts stating whether any buttonless characteristic was found.

bihanssen, 2017-10-02:
Closing due to inactivity and not able to reproduce.

closed 2017-10-02 by bihanssen