NordicSemiconductor/pc-ble-driver-js #67: pc-ble-driver subrepo not in release artifacts

IvanSanchez, 2017-08-04:


IvanSanchez, 2017-08-22:
This is alleviated by https://github.com/NordicSemiconductor/pc-ble-driver-js/commit/b8feb9335a38ddb373f2b546b8c696060f86f598 by providing a clearer message when `npm install`ing a freshly checked-out repo, but `npm install pc-ble-driver` is still an issue.

mrodem, 2017-10-20:
The submodule is fetched when when installing with npm. However, yarn does not fetch submodules, ref. https://github.com/yarnpkg/yarn/issues/1488.

Because of this it is unfortunately not possible to use yarn when depending on a Git repository that uses submodules, like pc-ble-driver-js. Yarn users need to either apply one of the workarounds proposed in the given issue or use npm.

closed 2017-10-20 by mrodem