NordicSemiconductor/pc-ble-driver-js #72: Cannot compile pc-ble-driver-js on Ubuntu 16.06

coderbyheart, 2017-09-21:
Using a [clean Ubuntu 16.06 Image from AWS](https://aws.amazon.com/marketplace/pp/B01JBL2M0O?ref=cns_srchrow) the compilation fails:

    [ 24%] Linking CXX shared library Release/pc-ble-driver-js-sd_api_v3.node
    /usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_thread.a(thread.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a shared object; recompile with -fPIC
    /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_thread.a: error adding symbols: Bad value
    collect2: error: ld returned 1 exit status
    CMakeFiles/pc-ble-driver-js-sd_api_v3.dir/build.make:336: recipe for target 'Release/pc-ble-driver-js-sd_api_v3.node' failed

<details>
  <summary>instance configuration</summary>
    sudo apt-get update -y > /dev/null
    sudo apt-get install -y unattended-upgrades apt-listchanges
    sudo unattended-upgrade
    sudo apt-get install -y build-essential cmake libboost-all-dev libudev-dev > /dev/null
</details>

<details>
 <summary>compile log</summary>
    > pc-ble-driver-js@2.0.0 install /home/ubuntu/workspace/gateway-common/node_modules/pc-ble-driver-js
    > node build.js

    info DIST Downloading distribution files.
    http DIST 	- http://nodejs.org/dist/v8.5.0/SHASUMS256.txt
    http DIST 	- http://nodejs.org/dist/v8.5.0/node-v8.5.0-headers.tar.gz
    info CMD CLEAN
    info RUN cmake -E remove_directory "/home/ubuntu/workspace/gateway-common/node_modules/pc-ble-driver-js/build"
    info CMD CONFIGURE
    WARN invalid config loglevel="notice"
    info RUN cmake "/home/ubuntu/workspace/gateway-common/node_modules/pc-ble-driver-js" --no-warn-unused-cli -DCMAKE_BUILD_TYPE="Release" -DCMAKE_LIBRARY_OUTPUT_DIRECTORY="/home/ubuntu/workspace/gateway-common/node_modules/pc-ble-driver-js/build/Release" -DCMAKE_JS_INC="/home/ubuntu/.cmake-js/node-x64/v8.5.0/include/node;/home/ubuntu/workspace/gateway-common/node_modules/nan" -DNODE_RUNTIME="node" -DNODE_RUNTIMEVERSION="8.5.0" -DNODE_ARCH="x64" -DCMAKE_CXX_FLAGS="-std=c++11" -DBOOST_ROOT="/usr/include/boost"
    Not searching for unused variables given on the command line.
    -- The C compiler identification is GNU 5.4.0
    -- The CXX compiler identification is GNU 5.4.0
    -- Check for working C compiler: /usr/bin/cc
    -- Check for working C compiler: /usr/bin/cc -- works
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Detecting C compile features
    -- Detecting C compile features - done
    -- Check for working CXX compiler: /usr/bin/c++
    -- Check for working CXX compiler: /usr/bin/c++ -- works
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- Detecting CXX compile features
    -- Detecting CXX compile features - done
    -- Architecture not set, using native 64-bit toolchain.
    -- Building with build type: Release.
    -- Looking for pthread.h
    -- Looking for pthread.h - found
    -- Looking for pthread_create
    -- Looking for pthread_create - not found
    -- Looking for pthread_create in pthreads
    -- Looking for pthread_create in pthreads - not found
    -- Looking for pthread_create in pthread
    -- Looking for pthread_create in pthread - found
    -- Found Threads: TRUE  
    -- Boost version: 1.58.0
    -- Found the following Boost libraries:
    --   thread
    --   system
    --   regex
    --   date_time
    --   chrono
    --   atomic
    -- Found Git: /usr/bin/git (found version "2.7.4") 
    -- Configuring done
    -- Generating done
    -- Build files have been written to: /home/ubuntu/workspace/gateway-common/node_modules/pc-ble-driver-js/build
    info CMD BUILD
    info RUN cmake --build "/home/ubuntu/workspace/gateway-common/node_modules/pc-ble-driver-js/build" --config Release
    Scanning dependencies of target pc_ble_driver_obj_sd_api_v3
[... 758 lines cut ...]
    [ 19%] Building CXX object CMakeFiles/pc-ble-driver-js-sd_api_v3.dir/src/serialadapter.cpp.o
    [ 20%] Building CXX object CMakeFiles/pc-ble-driver-js-sd_api_v3.dir/src/common.cpp.o
    [ 20%] Building CXX object CMakeFiles/pc-ble-driver-js-sd_api_v3.dir/src/driver.cpp.o
    [ 21%] Building CXX object CMakeFiles/pc-ble-driver-js-sd_api_v3.dir/src/driver_gap.cpp.o
    [ 21%] Building CXX object CMakeFiles/pc-ble-driver-js-sd_api_v3.dir/src/driver_gatt.cpp.o
    [ 22%] Building CXX object CMakeFiles/pc-ble-driver-js-sd_api_v3.dir/src/driver_gattc.cpp.o
    /home/ubuntu/workspace/gateway-common/node_modules/pc-ble-driver-js/src/driver_gattc.cpp: In member function ‘virtual v8::Local<v8::Object> GattcCharacteristicValueReadByUUIDEvent::ToJs()’:
    /home/ubuntu/workspace/gateway-common/node_modules/pc-ble-driver-js/src/driver_gattc.cpp:315:116: note: #pragma message: Support for GattcCharacteristicValueReadByUUIDEvent not implemented in AddOn for SDv3 and higher.
    #pragma message("Support for GattcCharacteristicValueReadByUUIDEvent not implemented in AddOn for SDv3 and higher.")
                                                                                                                        ^
    [ 22%] Building CXX object CMakeFiles/pc-ble-driver-js-sd_api_v3.dir/src/driver_gatts.cpp.o
    [ 23%] Building CXX object CMakeFiles/pc-ble-driver-js-sd_api_v3.dir/src/driver_uecc.cpp.o
    [ 23%] Building CXX object CMakeFiles/pc-ble-driver-js-sd_api_v3.dir/src/uECC/uECC.c.o
    [ 24%] Linking CXX shared library Release/pc-ble-driver-js-sd_api_v3.node
    /usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_thread.a(thread.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a shared object; recompile with -fPIC
    /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_thread.a: error adding symbols: Bad value
    collect2: error: ld returned 1 exit status
    CMakeFiles/pc-ble-driver-js-sd_api_v3.dir/build.make:336: recipe for target 'Release/pc-ble-driver-js-sd_api_v3.node' failed
    make[2]: *** [Release/pc-ble-driver-js-sd_api_v3.node] Error 1
    CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/pc-ble-driver-js-sd_api_v3.dir/all' failed
    make[1]: *** [CMakeFiles/pc-ble-driver-js-sd_api_v3.dir/all] Error 2
    Makefile:83: recipe for target 'all' failed
    make: *** [all] Error 2
    ERR! OMG Process terminated: 2
</details>

mrodem, 2017-09-21:
I believe you need to compile Boost yourself with the `-fPIC` flag, as described here: https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/Installation.md#building-boost.

I struggled with this myself a while ago, so I created an Ansible playbook that does provisioning of Node.js, Boost, pc-ble-driver-js, etc on Ubuntu 16.04: https://github.com/mrodem/pc-ble-linux-dev-env/tree/master/provision.d.

coderbyheart, 2017-09-21:
Thanks for getting back, I'll try that!

coderbyheart, 2017-09-21:
Compiling boost according to the provided Ansible playbook fixes the problem.

closed 2017-09-21 by coderbyheart