NordicSemiconductor/pc-ble-driver-js #88: Make device connection timeout not to be an error

kenr, 2017-10-07:
I think it would be better if device connection timeout did not generate an error in the callback.
The error should be reserved for other events like communication error with the adapter.

https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/749089aa7c0cfbb240adc134fb0844c0fd8f3ae3/api/adapter.js#L1058

bihanssen, 2017-10-09:
The timeout should be reported somewhere, or don't you think it's important info? One could argue that the absence of the connected-event is enough info that the connection didn't occur.

kenr, 2017-10-09:
device connection timeout is an event that is sent:
https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/749089aa7c0cfbb240adc134fb0844c0fd8f3ae3/api/adapter.js#L1054

bihanssen, 2017-10-09:
I'm quite sure we could remove the error callback on timeout in connect without causing problems.

mrodem, 2017-11-24:
Done in #111.

closed 2017-11-24 by mrodem