NordicSemiconductor/pc-ble-driver-js #90: On open error, no cleanup of resources is done

kenr, 2017-10-09:
On open error, no cleanup is done:
https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/749089aa7c0cfbb240adc134fb0844c0fd8f3ae3/src/driver.cpp#L717

mrodem, 2017-10-24:
Could you provide some more info regarding what needs to be cleaned up? If there is an error in open, then `error_code != NRF_SUCCESS`, and then `Adapter::AfterOpen()` will call `baton->mainObject->cleanUpV8Resources()`:

https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/40f27ed2633bbcf14a4239b753deab57ce8bd0ba/src/driver.cpp#L753

kenr, 2017-10-24:
On sd_rpc_open fail it only cleans up V8 resources:
https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/eb586cf2e653ded4c2431b8f93263d4777d808c1/src/adapter.cpp#L234

But the resources allocated here:
https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/40f27ed2633bbcf14a4239b753deab57ce8bd0ba/src/driver.cpp#L684

is not cleaned up.

closed 2017-10-31 by mrodem