NordicSemiconductor/pc-ble-driver-js #92: Inconsistent behavior of enableBLE

mrodem, 2017-10-09:
When calling `adapter.open()`, it is possible to specify `enableBLE: true` which then enables the BLE stack.   It is also possible to call `adapter.enableBLE()` after the adapter has been opened.

One would expect that `enableBLE: true` and `adapter.enableBLE()` does the same thing, but that is apparently not the case, ref. #89. Should investigate why these behave differently, and change it so that the behavior is consistent.

mrodem, 2017-10-09:
When specifying `enableBLE: true` in `adapter.open()`, the private `Adapter::enableBLE` function is called in driver.cpp. This sets a few default options: https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/749089aa7c0cfbb240adc134fb0844c0fd8f3ae3/src/driver.cpp#L496-L501

However, when calling `adapter.enableBLE(null, callback)` it uses the defaults from adapter.js which is a bit different: https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/964894267752e88dacbf0700c097b07a47a46e5d/api/adapter.js#L498-L515

Modifying the default behavior for either `enableBLE: true` or `adapter.enableBLE()` would be a breaking change, so I suppose it needs to wait until the next major version. Then I think it would be best to specify the default options in one place, so that it doesn't get out of sync.

bihanssen, 2017-12-11:
Resolved with PR https://github.com/NordicSemiconductor/pc-ble-driver-js/pull/119

closed 2017-12-11 by bihanssen