NordicSemiconductor/pc-ble-driver-js #100: unexpected behavior in att mtu handling

lemrey, 2017-11-06:
Hi, 

it seems that when an ATT MTU request is received, upon starting DFU the operation fails when it is unable to use 247 as ATT MTU. I would expect it to work regardless of the MTU set by the server.

bihanssen, 2017-11-06:
Thanks for reporting, we will look into this.

mrodem, 2017-11-10:
The DFU procedure in pc-ble-driver-js currently always sends an ATT MTU request of 247. The intention behind this was to use the highest MTU that is supported by the peripheral. However, if the peripheral also sends an ATT MTU request, then the values may conflict, which can result in an NRF_ERROR_INVALID_PARAM (0x07) error.

We will modify the DFU implementation in pc-ble-driver-js, so that it supports receiving ATT MTU requests from the peripheral.

mrodem, 2017-11-14:
The issue has been discussed with the reporter. We agreed that it is easiest if the peripheral just responds to the ATT MTU request from the central. This way we avoid having ATT MTU requests in both directions.

closed 2017-11-14 by mrodem