NordicSemiconductor/pc-ble-driver-js #118: Get RSSI during connection

rosek86, 2017-12-01:
Hi,

Is there any way to get RSSI during connection? I was trying something like this but this:

>      (this.adapter as any)._adapter.gapGetRSSI(this.peripheral.connectionHandle, (a, b) => {
>        this.logger.info(a, b);
>      });

but this generates this error:

>      undefined Error: Error occured when getting rssi. Errorcode: NRF_ERROR_INTERNAL (0x3)
>      logMessage: Failed to receive response for command
>      status: {"id":4,"name":"Unknown value","message":"Error sending packet to target. Code #3","time":"2017-04-14T09:50:47.269Z"}

It seems also that the error is second parameter in gapGetRSSI callback.

Kind Regards,
Krzysztof

bihanssen, 2017-12-21:
Hi,
the way you are calling the function is correct and should work, but there seems to be a bug in the underlying function implementation somewhere. We'll have to look into it.

bihanssen, 2017-12-21:
There is a bug in the connectivity firmware project for sdk 11.0. The SD_BLE_GAP_RSSI_GET function is missing from an array in middleware (components\serialization\connectivity\codecs\s130\middleware\conn_mw_items.c), so the connectivity fails during runtime when it receives a serialized function it doesn't have handler for. Adding a line for rssi_get in conn_mw_items.c seems to fix the problem.

Note: the function also seems to work out-of-the-box for SD API v3 (sdk 12).

Note2: Remember to call _adapter.gapStartRSSI first.

NarainMassand, 2020-09-08:
@bihanssen I am running into a similar error where I see NRF_ERROR_NOT_FOUND (0x5). Adding which line in conn_mw_items.c might fix it? Thanks!