NordicSemiconductor/pc-ble-driver-js #123: Error on iOS: "Failed to call gattsExchangeMtuReply"

Looveh, 2018-01-08:
When running the example script `examples/heart_rate_monitor.js` I can connect to the SDK using the nRF Connect app on Android without issues, but when I attempt to connect to the same SDK with nRF Connect on iOS I get the following error:

```
Searching for connected adapters...
onAdded: Adapter added: 000682402200.
Found the following adapters:
000682402200
Connected to adapter: 000682402200.
Opening adapter with ID: 000682402200 and baud rate: 115200...
#BLE_PERIPHERAL:  logMessage: Successfully opened /dev/cu.usbmodem1421. Baud rate: 115200. Flow control: none. Parity: none.
.
#BLE_PERIPHERAL:  status: {
 "id": 6,
 "name": "RESET_PERFORMED",
 "message": "Target Reset performed",
 "time": "2018-01-08T11:54:48.545Z"
}.
#BLE_PERIPHERAL:  status: {
 "id": 7,
 "name": "CONNECTION_ACTIVE",
 "message": "Connection active",
 "time": "2018-01-08T11:54:49.731Z"
}.
Opened adapter.
Initializing the heart rate service and its characteristics/descriptors...
Initialized the heart rate service and its characteristics/descriptors.
Setting advertisement data...
Set advertisement data.
Starting advertising...
Started advertising.
#BLE_PERIPHERAL:  logMessage: GAP_EVT_CONNECTED time:2018-01-08T11:54:54.559Z connHandle:0 peerAddr:[address:70:8E:13:DC:DA:A6 type:randomPrivateResolvable addrIdPeer:0] role:periph connParams:[minConnInterval:30 maxConnInterval:30 slaveLatency:0 connSupTimeout:720].
#BLE_PERIPHERAL:  deviceConnected: 70:8E:13:DC:DA:A6.
#BLE_PERIPHERAL:  descriptorValueChanged: {"_instanceId":"local.server.2.2.2","_characteristicInstanceId":"local.server.2.2","uuid":"2902","handle":14,"value":{"70:8E:13:DC:DA:A6.0":[0,0]},"readPerm":["open"],"writePerm":["open"],"variableLength":false,"maxLength":2}.
Disabling notifications on heart rate measurement characteristic...
#BLE_PERIPHERAL:  logMessage: GATTS_EVT_EXCHANGE_MTU_REQUEST time:2018-01-08T11:54:54.783Z connHandle:0 clientRxMtu:185.
#BLE_PERIPHERAL:  error: {
 "message": "Failed to call gattsExchangeMtuReply",
 "description": {
  "errno": 7,
  "errcode": "NRF_ERROR_INVALID_PARAM",
  "erroperation": "replying MTU exchange",
  "errmsg": "Error occured when replying MTU exchange. Errorcode: NRF_ERROR_INVALID_PARAM (0x7)\n"
 }
}.
```

Looveh, 2018-01-12:
Any idea on this? Is anybody able to connect using an iOS device?

bihanssen, 2018-01-12:
Hi, I assume you are building from master, and using latest connectivity firmware?

Try disabling JLink MSD on the devkit to see if it's related to packet truncation on MacOS (http://www.nordicsemi.com/nRFConnectOSXfix).

bihanssen, 2018-01-12:
I'm not able to reproduce the issue, it works for me running on mac from master branch, . Tried with connectivity v1.2.1 and v1.2.2.

bihanssen, 2018-11-13:
Closing due to inactivity. If this is still an issue, please re-post on [DevZone](https://devzone.nordicsemi.com).

closed 2018-11-13 by bihanssen