NordicSemiconductor/pc-ble-driver-js #127: Error during importing module from lib: Module did not self-register.

pokornyib, 2018-01-11:
I'm trying to use this Node.js lib in an electron app for firmware updating. 
The installations were successful (npm install pc-ble-driver-js --save), anything rebuilded with msvc-14.0 and Boost 1.65.0 under Windows 10 64bit. But when I try to import something from this lib, I receive this error:
```
Uncaught Error: Module did not self-register.
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:503:32)
    at tryModuleLoad (module.js:466:12)
    at Function.Module._load (module.js:458:3)
    at Module.require (module.js:513:17)
    at require (internal/module.js:11:18)
    at bindings (C:\SandBox\EdgeApp\node_modules\bindings\bindings.js:83:44)
    at Object.<anonymous> (C:\SandBox\EdgeApp\node_modules\pc-ble-driver-js\api\adapterFactory.js:41:41)
```
Can anyone help form me, to solve this issue?

mrodem, 2018-01-11:
I suspect that the module was not compiled against Electron's headers when you did `npm install pc-ble-driver-js`. There are various ways of doing this, as described here: https://github.com/electron/electron/blob/master/docs/tutorial/using-native-node-modules.md

pokornyib, 2018-01-11:
If I try rebuild pc-ble-driver-js manually with `node-gyp rebuild --target=1.2.3 --arch=x64 --dist-url=https://atom.io/download/electron` , then I receive the following message: `gyp: binding.gyp not found (cwd: C:\SandBox\EdgeApp\node_modules\pc-ble-driver-js) while trying to load binding.gyp`

There is no binding.gyp file in pc-ble-diver-js folder. What can I do?

pokornyib, 2018-01-11:
@mrodem Can you send me an prebuild version for electron 1.8.1?

mrodem, 2018-01-11:
Try adding this to .npmrc in your project directory:

```
runtime = electron
target = 1.8.1
disturl = https://atom.io/download/atom-shell
```

Then re-run `npm install pc-ble-driver-js`. It should now pull down the headers for Electron 1.8.1 and compile against those.

pokornyib, 2018-01-11:
Thanks! It helped, now the pc-ble-driver-js works. But now an other npm package is not working. What does this file with this lines?

mrodem, 2018-01-12:
The `runtime`, `target`, and `disturl` values default to your currently installed Node.js version, meaning that native modules will be built to work with your Node.js version. However, Electron usually comes with a different V8 (Chrome JS engine) version than the one provided by Node.js, so native modules that are built for your Node.js version will not work in Electron.

You have to specify that you want to build the native modules for Electron. Adding configuration to .npmrc is just one way to do this. Alternative solutions (environment variables, electron-rebuild, etc.) are described in https://github.com/electron/electron/blob/master/docs/tutorial/using-native-node-modules.md. If .npmrc is causing problems for other packages, perhaps you could try one of the alternative solutions.

pokornyib, 2018-01-12:
Now I understand how it works, but when I trying building with `electron-rebuild`, then nothing happens. Just say 'Rebuild complete'. With `node-gyp rebuild` fails, because in `pc-ble-driver-js` folder didn't contains `binding.gyp` file.

It only works with npm building, with setting these environment variables, which you mentioned. 
Now the app works, but only if I install everything, but this package without .npmrc, and then install `pc-ble-driver-js` with .npmrc.

Is there a better way?

mrodem, 2018-01-12:
This project is using cmake-js instead of node-gyp, and it appears that electron-rebuild depends on node-gyp, so that will not work.

I do not know which other packages you are depending on, but adding runtime, target, and disturl in .npmrc is a common way to specify that you want to build for Electron, and should normally be supported. See f.ex. http://www.nodegit.org/guides/install/electron/ and https://github.com/electron-userland/electron-spellchecker/blob/master/.npmrc. Perhaps you could look into why your other package(s) are failing when runtime, target, and disturl are configured?

pokornyib, 2018-01-12:
I use an own npm package, which uses no native libs, expect serialport. But the problem is not with serialport, but with tslint. On the console comes up no error, just stops the app on line `... =require("tslint/lib/error")` . (The `console.log('...')` before this line works, after this not.) 
Tslint is native lib? Or what is happening? 

Thank you for this lot of help!

closed 2018-02-07 by pokornyib