NordicSemiconductor/pc-ble-driver-js #148: query - "This will work as long as precompiled binaries exist for your platform/runtime environment, ref. the files attached to the Releases"

tonykappen, 2018-03-27:
Hi, 

I am a newbie to embedded development. I am well versed with java script. I am under the impression nordic allows you to communicate with java script to the development kit. I have a nRF52 (nRF52832) development kit (PCA10040). I am trying to installing the pc-ble-driver-js from the link https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/master/Installation.md. I wish to develop the javascript application with electron. I didnot understand the portion of the text "This will work as long as precompiled binaries exist for your platform/runtime environment, ref. the files attached to the Releases". Can you please give me further explanation on the meaning of the line. I am developing on Windows 10 OS with a 32bit version of node installed. Your help is much appreciated. 

Sincerely,

Tony.

tonykappen, 2018-03-28:
Thank you for helping me out. I think pc-ble-driver-js got installed. When I tried to run heart rate monitor example I got the error as shown in the below screen shot
[image]

Can you please help me out?

bencefr, 2018-04-03:
Hi @tonykappen ,
Sorry for the delay, Easter in Norway is really long :)
Judging by the picture you have an incorrect firmware on the device, meaning one that runs with different baudrate than expected. I might be wrong, but it doesn't hurt to try to either modify the example source to match the baudrate of the firmware or flash the correct firmware.
You have the choice of 115200 and 1000000, follow the guide [here](https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/Installation.md#hardware-setup).

tonykappen, 2018-04-05:
I also tried flashing the pc-ble-driver with nRFgo studio. I am trying to flash the connectivity firmware using the program softdevice tab.
[image]
I am getting the error as shown in the above screen shot. Next, I try to flash the connectivity firmware using Program Application tab. Now the nRF52 dev kit is getting programmed. Please refer the below screenshot. 
[image]
Please refer the last line of the log file. I assume that that the connectivity hex file got flashed as an application firmware. After this, I flash the application firmware in the following path “nRF5_SDK_14.2.0_17b948a\examples\ble_peripheral\ble_app_hrs\pca10040\s132\ses\Output\Debug\Exe”, the development kit gets flashed. Please refer the below screenshot. 
[image]
The PC gets connected for some time and then it goes back the paired state. After that the PC shows that the development kit goes back to the paired state. When I run the node command, I get the following error. Please refer the below screenshot. 

[image]

bihanssen, 2018-11-13:
Closing this issue. If you still need assistance or are having problems, please re-post on [DevZone](https://devzone.nordicsemi.com).

closed 2018-11-13 by bihanssen