NordicSemiconductor/pc-ble-driver-js #194: TX complete event not present in soft device v5

pambardekar, 2018-09-11:
We have been relying on TX complete event with soft device v3. This event is missing from v5. Is there another mechanism to get notified about completion of a write operation?

kenr, 2018-09-11:
Hi @pambardekar,
What is your setup? How are you using SDv5 in pc-ble-driver-js?

pambardekar, 2018-09-12:
I am using 32 bit node js version 9.4.0. I am looking to write a node.js addon on top of pc-dle-driver.

pambardekar, 2018-09-12:
Dev platform is Windows 10 64 bit.

pambardekar, 2018-09-14:
Hi @kenr 

Did you get a chance to look into this?

kenr, 2018-09-14:
@pambardekar thanks for the info. I wanted to check if you are using pc-ble-driver-js and not pc-ble-driver. The support for SoftDevice v5 is not complete yet. The latest released version (v2.4.2) of pc-ble-driver-js does not support SoftDevice v5, only SoftDevice v2 and v3.

However SoftDevice v5 supports TX_COMPLETE: http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s132.api.v5.0.0%2Fgroup___b_l_e___g_a_t_t_c___f_u_n_c_t_i_o_n_s.html&anchor=ga90298b8dcd8bbe7bbe69d362d1133378

pambardekar, 2018-09-14:
@kenr Thanks for the response. I understand that I cannot use pc-ble-driver-js to communicate with SoftDevice v5. I am investigating if I can write my own node JS addon that integrates with pc-ble-driver that supports SoftDevice v5 (master).

pambardekar, 2018-09-25:
I was able to locate details correctly. Thanks.

closed 2018-09-25 by pambardekar

NarainMassand, 2020-10-22:
Hi @pambardekar , What ended up working for you? I am trying to figure out a way to check if the packet has been received. Thanks!