NordicSemiconductor/pc-ble-driver-js #195: Support for Raspberry Pi 3

lbabaly, 2018-10-02:
Hello,

Do you plan to support Raspberry Pi 3 platform with pc-ble-driver-js?
I can't install the package because the compilation of the pc-nrfjprog-js fails. Is there a workaround for that?
The usb npm package also fails during installation.
I'm using node version 10.11 on my RPI.

Reagrds,
Laszlo

bencefr, 2018-10-03:
Hi @lbabaly ,
Please try installing with node version 8.x LTS, until all required dependencies are updated to support node 10.

lbabaly, 2018-10-03:
Hello Bence,

Thank you for your support. I managed to install with node version 8.x but there were two important steps that were necessary to do before the installation:

1.) Boost libraries must be installed & compiled. Environment variable must be set properly pointing to the installation directory:
`export BOOST_ROOT=~/boost/boost_1_xx_y/`

2.) nan npm package must be installed manually before installing pc-ble-driver-js otherwise the compilation get stucked.

Regards,
Laszlo