NordicSemiconductor/pc-ble-driver-js #197: Unable to build the latest package

jayakanth459, 2018-10-17:
Hi,

I'm unable to build the package after clone.

I still some package issues. could you please help me on this.

Please find the below log after git clone.

C:\MyPractice>npm install

> pc-nrfjprog-js@1.2.0 install C:\MyPractice\node_modules\nrf-device-setup\node_modules\pc-nrfjprog-js
> node-pre-gyp install --fallback-to-build=false || node build.js

node-pre-gyp ERR! install error
node-pre-gyp ERR! stack Error: 404 status code downloading tarball https://github.com/NordicSemiconductor/pc-nrfjprog-js/releases/download/v1.2.0/pc-nrfjprog-js-v1.2.0-node-v57-win32-x64.tar.gz
node-pre-gyp ERR! stack     at Request.<anonymous> (C:\MyPractice\node_modules\node-pre-gyp\lib\install.js:118:27)
node-pre-gyp ERR! stack     at emitOne (events.js:121:20)
node-pre-gyp ERR! stack     at Request.emit (events.js:211:7)
node-pre-gyp ERR! stack     at Request.onRequestResponse (C:\MyPractice\node_modules\node-pre-gyp\node_modules\request\request.js:1074:10)
node-pre-gyp ERR! stack     at emitOne (events.js:116:13)
node-pre-gyp ERR! stack     at ClientRequest.emit (events.js:211:7)
node-pre-gyp ERR! stack     at HTTPParser.parserOnIncomingClient [as onIncoming] (_http_client.js:543:21)
node-pre-gyp ERR! stack     at HTTPParser.parserOnHeadersComplete (_http_common.js:112:17)
node-pre-gyp ERR! stack     at TLSSocket.socketOnData (_http_client.js:440:20)
node-pre-gyp ERR! stack     at emitOne (events.js:116:13)
node-pre-gyp ERR! System Windows_NT 10.0.15063
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\MyPractice\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build=false"
node-pre-gyp ERR! cwd C:\MyPractice\node_modules\nrf-device-setup\node_modules\pc-nrfjprog-js
node-pre-gyp ERR! node -v v8.11.4
node-pre-gyp ERR! node-pre-gyp -v v0.6.39
node-pre-gyp ERR! not ok
404 status code downloading tarball https://github.com/NordicSemiconductor/pc-nrfjprog-js/releases/download/v1.2.0/pc-nrfjprog-js-v1.2.0-node-v57-win32-x64.tar.gz
info DIST Downloading distribution files.
http DIST       - http://nodejs.org/dist/v8.11.4/SHASUMS256.txt
http DIST       - http://nodejs.org/dist/v8.11.4/win-x64/node.lib
http DIST       - http://nodejs.org/dist/v8.11.4/node-v8.11.4-headers.tar.gz
info CMD CLEAN
info RUN cmake -E remove_directory "C:\MyPractice\node_modules\nrf-device-setup\node_modules\pc-nrfjprog-js\build"
info CMD CONFIGURE
WARN invalid config loglevel="notice"

info RUN cmake "C:\MyPractice\node_modules\nrf-device-setup\node_modules\pc-nrfjprog-js" --no-warn-unused-cli -DCMAKE_BUILD_TYPE="Release" -DCMAKE_JS_INC="C:\MyPractice\.cmake-js\node-x64\v8.11.4\include\node;C:\MyPractice\node_modules\nan" -DNODE_RUNTIME="node" -DNODE_RUNTIMEVERSION="8.11.4" -DNODE_ARCH="x64" -DCMAKE_JS_LIB="C:\MyPractice\.cmake-js\node-x64\v8.11.4\win-x64\node.lib"
Not searching for unused variables given on the command line.
-- Building for: NMake Makefiles
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:4 (project):
  The CMAKE_C_COMPILER:

    cl

  is not a full path and was not found in the PATH.

  To use the NMake generator with Visual C++, cmake must be run from a shell
  that can use the compiler cl from the command line.  This environment is
  unable to invoke the cl compiler.  To fix this problem, run cmake from the
  Visual Studio Command Prompt (vcvarsall.bat).

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:4 (project):
  The CMAKE_CXX_COMPILER:

    cl

  is not a full path and was not found in the PATH.

  To use the NMake generator with Visual C++, cmake must be run from a shell
  that can use the compiler cl from the command line.  This environment is
  unable to invoke the cl compiler.  To fix this problem, run cmake from the
  Visual Studio Command Prompt (vcvarsall.bat).

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "C:/MyPractice/node_modules/nrf-device-setup/node_modules/pc-nrfjprog-js/build/CMakeFiles/CMakeOutput.log".
See also "C:/MyPractice/node_modules/nrf-device-setup/node_modules/pc-nrfjprog-js/build/CMakeFiles/CMakeError.log".
ERR! OMG Process terminated: 1
Unhandled rejection Error: Process terminated: 1
    at ChildProcess.<anonymous> (C:\MyPractice\node_modules\cmake-js\lib\cMake.js:372:28)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)

> serialport@6.2.2 install C:\MyPractice\node_modules\serialport
> prebuild-install || node-gyp rebuild

> pc-nrfjprog-js@1.3.1 install C:\MyPractice\node_modules\pc-nrfjprog-js
> node scripts/pre-install.js && ( node-pre-gyp install --fallback-to-build=false || node build.js )

Unsupported architecture: x64. On Windows, the nrfjprog libraries currently require 32-bit Node.js.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.4 (node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! pc-nrfjprog-js@1.3.1 install: `node scripts/pre-install.js && ( node-pre-gyp install --fallback-to-build=false || node build.js )`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the pc-nrfjprog-js@1.3.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\MyPractice\AppData\Roaming\npm-cache\_logs\2018-10-17T06_55_08_509Z-debug.log

chunfantasy, 2018-10-17:
Hi @jayakanth459 the current latest package does not support Windows 64 bit.

chunfantasy, 2018-10-17:
@jayakanth459 if you only want to use it, you can download it from here https://github.com/NordicSemiconductor/pc-ble-driver-js/releases for your node js 32 bit.

If you somehow need to build from source, then it is not only about node js, it is also related to your compiler. You can follow the instruction here https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/master/Installation.md

Note, As building pc-ble-driver-js also involves building pc-ble-driver, you first need to follow the pc-ble-driver installation instructions. https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/Installation.md

jayakanth459, 2018-10-17:
Thanks chun. I've cloned "https://github.com/NordicSemiconductor/pc-ble-driver-js" and did npm install. it doesn't show any exceptions now after git cloneing.

Let say i want to execute a LED blink of nordic using pc-ble-driver-js connecting to the device. is there any ready made sample available which I can try and check it.

chunfantasy, 2018-10-17:
I suggest you to take a look at our examples
https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/master/examples/heart_rate_collector.js
https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/master/examples/heart_rate_monitor.js
This may guide you how to use pc-ble-driver-js.

And welcome to ask questions at https://devzone.nordicsemi.com/.

Since this is not an issue, I close it.

closed 2018-10-17 by chunfantasy