NordicSemiconductor/pc-ble-driver-js #214: Unable to use pc-ble-driver-js with electron 4.0.8

pambardekar, 2019-03-22:
We have been using pc-ble-driver-js 2.4.3 with electon 1.8.1. With the latest 2.5.1 release, I am trying to upgrade to electron 4.0.8. When loading the pc-ble-driver-js node module I am getting following error:

XXXX\node_modules\electron-log\renderer.js:34 [10:14:51.109] Failed to add module. Error: Error: The specified procedure could not be found.
\\?\XXXXX\node_modules\pc-ble-driver-js\build\Release\pc-ble-driver-js-sd_api_v2.node

This error generally indicates that a required DLL is missing. I have used dependency walker to check the dependencies but I couldn't nail down the error.

I have Visual Studio 2015 installed. I also have VC++ 2015 redistributable packages installed.

Is there any other dependency that needs to be installed?

pambardekar, 2019-03-25:
I had to make a few changes to pc-ble-driver-js build for it to work with electron 4. I referred to https://github.com/nodejs/node-addon-api/issues/269#issuecomment-455580129. I am wondering how other users are getting it to work.

bihanssen, 2019-03-25:
@pambardekar we have not taken the electron 4 prebuilt binaries into use internally yet, so it might be that there are some issues hidden there. 
We know there were issues regarding ABI mismatch in electron < v4.0.4, but as you are using 4.0.8 and the released binaries are v4.0.6, that should not be a problem: https://github.com/electron/electron/pull/16687

christophgreen, 2019-05-21:
@bihanssen I am also trying to upgrade my application to Electron 4. Electron >=4.0.4 bumped the NMV to 69 and the NMV used to build the electron 4 binaries is 64. 

Do you have plans to publish updated binaries to support Electron >= 4.0.4?

bihanssen, 2019-05-22:
@christophgreen. the plan is to release a new version of pc-ble-driver-js within the next weeks. It will also include fix #216 related to electron 4.

christophgreen, 2019-06-18:
@bihanssen Any update on when this will be available?

bihanssen, 2019-06-19:
@christophgreen, some delays due to integration tests, will be out once resolved.

alwa-nordic, 2019-07-03:
There is a fix for this issue in v2.6.0. @christophgreen Can you confirm that it works for you?

christophgreen, 2019-07-03:
@alwa-nordic Yes we were able to get this running with electron 4.1. I'll open an issue if we encounter any other problems while testing the updated library.

closed 2019-07-04 by alwa-nordic

shaneeelliott, 2019-11-13:
Does anyone have a sample electron application implementing pc-ble-driver-js ?

bencefr, 2019-11-14:
@shaneeelliott , yes we do, our application is: https://github.com/NordicSemiconductor/pc-nrfconnect-core which contains pc-ble-driver-js, the module itself is used in the app that core loads: https://github.com/NordicSemiconductor/pc-nrfconnect-ble