NordicSemiconductor/pc-ble-driver-js #234: Installation problem

violetgee, 2019-11-29:
When trying to install the latest version, the following error appears:

`Validation of nrfjprog libraries failed: Could not locate the bindings file. Tried:
 → C:\Users\Documents\GitHub\pc-ble-driver-js\node_modules\pc-nrfjprog-js\build\pc-nrfjprog-js.node`

`node-pre-gyp ERR! stack Error: 404 status code downloading tarball https://github.com/NordicSemiconductor/pc-nrfjprog-js/releases/download/v1.6.0/pc-nrfjprog-js-v1.6.0-node-v72-win32-ia32.tar.gz`

I just downloaded the latest version and I'm trying to run the command npm install and npm install pc-ble-driver-js.  It also doesn't work:
     $ git clone https://github.com/NordicPlayground/vcpkg.git
    $ git checkout tags/2019-06-09
    $ ./vcpkg/bootstrap.bat
    $ ./vcpkg/vcpkg install nrf-ble-driver:x64-windows

chunfantasy, 2019-11-29:
@violetgee I suggest you to use node 8, 10 or 11 if you just wanna use it without modification since we do not have the release for node 12 support yet (I suppose you are using node 12). If you have to use node 12 then you need to build it yourself before we have node 12 support.

violetgee, 2019-11-29:
Thank you so much! Now it works but I have another quick question. I try to run test/simpleScan.test.js and I have this error: 
ReferenceError: jest is not defined

closed 2019-12-02 by violetgee