NordicSemiconductor/pc-ble-driver-js #236: Electron instalation

violetgee, 2019-12-03:
Hello,
I have a problem with the installation. First I run this command in main folder:
npm install -g electron-prebuilt

> These variables can be set in .npmrc file in the root of your Electron based project. Example .npmrc file content:
> runtime = Electron
> target = 1.16.6
I understand that I need to create a file with this extension, right?  And what to do next with this file?
Add the environment variables npm_config_runtime and npm_config_target (must be set for npm install.) Thats ok.
But what with this?:
The variables should be "Electron" and the Electron version, respectivly.

datenreisender, 2019-12-03:
> Hello,
> I have a problem with the installation. First I run this command in main folder:
> npm install -g electron-prebuilt
>
>
> I understand that I need to create a file with this extension, right?

No, the file name does not just need to have this extension, the whole filename should be `.npmrc`.

Based on your description, I have the impression that you seem quite inexperienced with using npm and building JavaScript/Electron apps. You might need to build up some basic knowledge about that first.

violetgee, 2019-12-03:
I have already built applications in electron and javascript.
I only have problem adding electron to this project.

datenreisender, 2019-12-03:
What do you mean with “this project”? `pc-ble-driver-js` or another project using `pc-ble-driver-js`? 

Did creating a [`.npmrc`](https://docs.npmjs.com/files/npmrc) help? 

By the way: `electron-prebuilt` is deprecated, you should not use it anymore.

violetgee, 2019-12-03:
Ok, thanks for advice :)
I created a new empty electron project and then add there pc-ble-driver-js and thats helped.
Only problem I have now is that I cant debug because I have this error:

> Error: Cannot find module 'electron'

on 
`const` {app, BrowserWindow} = require('electron')

violetgee, 2019-12-03:
Ok I forgot to add this in launch.json eh

`{
    "version": "0.2.0",
    "configurations": [
      {
        "name": "Debug Main Process",
        "type": "node",
        "request": "launch",
        "cwd": "${workspaceFolder}",
        "runtimeExecutable": "${workspaceFolder}/node_modules/.bin/electron",
        "windows": {
          "runtimeExecutable": "${workspaceFolder}/node_modules/.bin/electron.cmd"
        },
        "args" : ["."],
        "outputCapture": "std"
      }
    ]
  }`

Thank you @datenreisender datenreisender for help!

closed 2019-12-03 by violetgee