NordicSemiconductor/pc-ble-driver-js #240: pc-ble-driver-js NODE_MODULE_VERSION Error

violetgee, 2019-12-18:
Hello,

I try to run electron app with pc-ble-driver-js and I have this error:

Error: The module '\\?\C:\app\build\Release\pc-ble-driver-js-sd_api_v2.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 67. This version of Node.js requires
NODE_MODULE_VERSION 69. Please try re-compiling or re-installing

electron:  v4.1.3

node: v11.15.0

pc-ble-driver-js: v2.6.2

Edit: tried with electron version 5.0.0, 6.0.0, 7.0.0 and the latest, and the error has not disappeared

How to reproduce:

1. Download source code and pc-ble-driver-js-sd_api_v3-v2.6.2-electron-v5.0-win32-x64.tar.gz from https://github.com/NordicSemiconductor/pc-ble-driver-js/releases

2. npm install electron@5.0.0 --save-dev 

3. change launch.json :

`{
    "version": "0.2.0",
    "configurations": [
      {
        "name": "Debug Main Process",
        "type": "node",
        "request": "launch",
        "cwd": "${workspaceFolder}",
        "runtimeExecutable": "${workspaceFolder}/node_modules/.bin/electron",
        "windows": {
          "runtimeExecutable": "${workspaceFolder}/node_modules/.bin/electron.cmd"
        },
        "args" : ["."],
        "outputCapture": "std"
      }
    ]
  }`

bencefr, 2019-12-18:
Do you have `.npmrc` in your source root? E.g.: https://github.com/NordicSemiconductor/pc-nrfconnect-launcher/blob/master/.npmrc

violetgee, 2019-12-18:
yes

violetgee, 2019-12-18:
Also when I run npm install pc-ble-driver-js I got this error:

>  node-pre-gyp install --fallback-to-build=false
> 
> node-pre-gyp WARN Using request for node-pre-gyp https download
> node-pre-gyp ERR! install error
> node-pre-gyp ERR! stack Error: Unknown Runtime: 'npm install'

bencefr, 2019-12-18:
Are your sources public?

violetgee, 2019-12-18:
no, but I don't change anything yet. I just try to add electron [https://electronjs.org/docs/tutorial/first-app#electron-development-in-a-nutshell] to pc-ble-driver-js

bencefr, 2019-12-18:
Well, you wouldn't add electron to pc-ble-driver-js, you would add pc-ble-driver-js to a project that is using electron.

violetgee, 2019-12-18:
Yeah I've tried to do it this way. The result is the same :/

bencefr, 2019-12-18:
Try these steps to create a new app (written in bash, since I don't speak windows commands):
```bash
$ mkdir app
$ cd app
$ npm init
$ npm install electron@5.0.6
$ echo -e "runtime = electron\ntarget = 5.0.6\ndisturl = https://electronjs.org/headers" > .npmrc
$ npm install pc-ble-driver-js
$ echo "<html></html>" > index.html
$ npx electron -i
```
in electron command line:
```js
new (require('electron')).BrowserWindow({webPreferences: {nodeIntegration: true}}).loadFile(`./index.html`);
```
This opens an electron window where you can open the devtools and switch to console. In that console:
```js
module.paths.push(process.cwd()+"/node_modules");
const bleDriver = require('pc-ble-driver-js');
```
Now I don't know why the local _node_modules_ needs to be added to `module.paths` here, but in a correctly webpacked environment that should not an issue. However these steps ensure that you have the correct module.

violetgee, 2019-12-18:
Still:
`Error: The module '\\?\c:\app\build\Release\pc-ble-driver-js-sd_api_v2.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 70. This version of Node.js requires
NODE_MODULE_VERSION 67. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).`

bencefr, 2019-12-18:
You must have something in your environment that effects which binding is downloaded.
If you don't create a clean directory please remove your _node_modules_ and _package-lock.json_.

I updated my previous comment since electron doesn't have REPL on windows.

violetgee, 2019-12-18:
ok now I have: 

> pc-ble-driver-js-sd_api_v2.node is not a valid Win32 application

violetgee, 2019-12-19:
I tried on another computer and the error still occurs :/

> pc-ble-driver-js-sd_api_v2.node is not a valid Win32 application

violetgee, 2019-12-19:
I added this:
`runtime = electron
target = 4.1.0
target_arch = x64
disturl = https://atom.io/download/atom-shell
progress = true`
to .npmrc file and it works! 
Thank you @bencefr  for help :)

closed 2019-12-19 by violetgee