NordicSemiconductor/pc-ble-driver-js #248: I want to know next release plan

mitoconcrete, 2020-04-16:
Hi Guys!
I using 2.6.2 release version now.
But this version isn't support over 8 connections. 
The man named "bencefr" who merge code when last year, November, said next release possible up to 20 connection.
So, I wait eagerly next release. 
Please tell me next release plan. 
Thanks.    

below link is merge comment.
[https://github.com/NordicSemiconductor/pc-ble-driver-js/pull/235](url)

bencefr, 2020-04-17:
Hi @mitoconcrete ,
I don't recall commenting anything about the number of connections.
However the limitation comes from the connectivity firmware which is built as part of the https://github.com/NordicSemiconductor/pc-ble-driver
We don't plan to increase the number of connections, since it most likely requires more memory on the nRF SoC.
The next release will be 2.7.0 in the near future with the features already available on master branch.

mitoconcrete, 2020-04-17:
Thanks @bencefr for quick reply!
For reference, I thought I could connect more than 8 by looking at the link below.
[https://devzone.nordicsemi.com/f/nordic-q-a/48502/error-for-connecting-more-than-4-centrals-to-multiperipheral-example-code/192560#192560](url)