NordicSemiconductor/pc-ble-driver-js #249: how do i maintain connect after call connect method?

mitoconcrete, 2020-04-17:
Hi!
I have a auto-disconnect issue. 
That issus is occured after called connect method. 
Connect is very very good working, but after about 40sec all kit disconnect automatically.
I think that is connectParameter setting problem. But I don't know how to set parameters for remain connection. 
Please Help me. 
Thanks!

Below code is my connect parameters setting.
```
 const scanParameters = {
      active: true,
      interval: 100,
      window: 50,
      timeout: 20,
    };
    const connParams = {
      min_conn_interval: 7.5,
      max_conn_interval: 7.5,
      slave_latency: 10,
      conn_sup_timeout: 4000,
    };
```

ShaharHD, 2021-03-17:
This still happens.
Mac OS, node v12

Any idea how to resolve this?

mitoconcrete, 2021-03-17:
nope i dont resolve this problem yet.
in Window 10, node v12.

ShaharHD, 2021-03-17:
I see the disconnect timeout around 30 seconds.
During those 30 seconds everything works - no idea what's causing this disconnect.

Maybe there's an undocumented "keep alive" somewhere in https://github.com/NordicSemiconductor/pc-nrfconnect-ble ?
...been looking there for clues.

ShaharHD, 2021-03-17:
@kenr @mitoconcrete I've created a sample project:
https://github.com/WeCanDoIT-Ltd/pc-ble-driver-js-test

which shows the exact issue by using the `ble_app_uart` from the `nrf5_sdk` examples.

It will connect, subscribe for the TX event - but will not receive notifications and will disconnect for some reason after 30 seconds or so.

I've verified subscription works by adding the following to `nus_data_handler` in `main.c` :

```c
    else if (p_evt->type == BLE_NUS_EVT_COMM_STARTED)
    {
        NRF_LOG_INFO("BLE_NUS_EVT_COMM_STARTED");
    }
    else if (p_evt->type == BLE_NUS_EVT_COMM_STOPPED)
    {
        NRF_LOG_INFO("BLE_NUS_EVT_COMM_STOPPED");
    }
```

ShaharHD, 2021-03-18:
Updated my test to try and follow https://devzone.nordicsemi.com/f/nordic-q-a/57789/connecting-to-device-using-pc-ble-driver-py-causes-disconnect-after-60-seconds

But it seems it's unrelated as changing `NEXT_CONN_PARAMS_UPDATE_DELAY` has not affect on the disconnect timeout

ShaharHD, 2021-03-19:
I think I resolved the issue - but it might varies with other projects.

In my case, I needed to add handling for the following events and **respond**
```js
  adapter.on('attMtuRequest', (device, newMtu) => {
    adapter.attMtuReply(device.instanceId, newMtu, (err, mtu) => {});
  });

  adapter.on('dataLengthUpdateRequest', (device, newOctets) => {
    adapter.dataLengthUpdate(device.instanceId, newOctets, () => {});
  });
```

If you're still getting disconnected, the fastest way IMHO to discover which requests you are not responding to is as follows:

- Open `node_modules/pc-ble-driver-js/api/adapter.js`
- Add in `_eventCallback(eventArray)` the following to filter which logs you want to *avoid* printing (otherwise it's overwhelming)

```js
        const filterLog = [
            this._bleDriver.BLE_GATTC_EVT_CHAR_DISC_RSP, 
            this._bleDriver.BLE_GAP_EVT_ADV_REPORT,
            this._bleDriver.BLE_GATTC_EVT_READ_RSP,
            this._bleDriver.BLE_GATTC_EVT_PRIM_SRVC_DISC_RSP,
        ];
```
And then add just below `this.emit('logMessage', logLevel.DEBUG, text.toString());` the following:

```js
            if (filterLog.includes(event.id) === false) {
                console.log(text.toString());
            }
```

And try and follow which events you are not responding to and getting disconnected or timed-out on (reason).