NordicSemiconductor/pc-ble-driver-js #259: Bluetooth Long Range Support

rosek86, 2020-08-26:
Hello,

We successfully use pc-ble-driver-js for a few years now, it works very well and till now it suited all our needs. We're now requested to incorporate Bluetooth Long Range in one of our next product.

It seems that pc-ble-driver-js has support for V5 now which is still s132 (no support for long-range). Do you have any plans to upgrade to s140 in the near future (pc-ble-driver V6)? Is there a roadmap available?

Kind regards,
Krzysztof

bihanssen, 2020-08-26:
Hi, @rosek86.
We have held back on v6 support, mainly due to poor scan performance when used over a serialized transport, due to a change in behavior of the scan API in SD v6.
Question, in your case, would degraded performance of scanning (and possibly loss of adv packets) be an issue?

rosek86, 2020-08-27:
Hi @bihanssen,

Thank you very much for your speedy reply.

Everything depends on the rate of adv packets loss. I don't think we have high constraints here, though. We modify pc-ble-driver to reject all adverts which don't belong to our products, and we keep ~200 ms adverts interval.

If there is a development version we'd be more than happy to give this a go.

bihanssen, 2020-08-27:
Hi,
there's no development version available currently of ble-driver-js with SD v6. However, regarding performance of scanning it would behave similarly to ble-driver (C/C++) if you would like to check it out immediately.

ShaharHD, 2022-05-22:
@rosek86 

I would take the pc-ble-driver-v6 and compare the changes between the h files of v5 and v6, branch out this repo and implement the missing headers/functions in adapter.js and the various other files.

Shahar

td-krzysiek, 2022-05-25:
Thanks, we would like to make an attempt with Bluetooth Long Range but it has lower priority at the moment and we can't embark upgrade to v6 ourself. It seems that Nordic Semiconductor abandon the project so it seems unlikely that v6 will be implemented by them anytime soon (I have couple pull requests hanging here for months now too).

I think there are two main approaches for our Node.js software, upgrade this repository to v6 at one point (with cooperation to yourself?) or migrate the firmware to Zephyr HCI controller and use e.g. [noble](https://github.com/abandonware/noble) with a facade layer to support API from this library.

For the second approach I also started working on altenative library which connects to Zephyr HCI controller but due to lack of time after work I can't support this recently too (https://github.com/rosek86/n-ble).

Kind regards